wellenvogel/esp32-nmea2000 #7: SeaTalk 1 connection

norbert-walter, 2021-11-25:
Implementation of SeaTalk 1 for M5Stack Atom with a special driver circuit  for SeaTalk with a ATOMIC DIY Proto Kit.

hwater, 2023-03-15:
Anyone here for an interrupt driven bitbangling serial 9 bit driver for core0?
If i remember correctly, the z80 made it in software with autobaudrating up to 57000 Baud.
For an ESP32 ppppfffft.!

hwater, 2023-03-16:
Just for fun i just asked ChatGPG https://chat.openai.com/chat this Q: program esp32 seatalk 9bit

hwater, 2023-03-16:
and got:
```c++
#include <HardwareSerial.h>
#include <Seatalk.h>

#define SERIAL_RX_PIN 2
#define SERIAL_TX_PIN 3

HardwareSerial Serial9bit(1);
Seatalk seatalk;

void setup() {
  Serial.begin(4800);
  Serial9bit.begin(4800, SERIAL_9BIT, SERIAL_8E1, SERIAL_RX_PIN, SERIAL_TX_PIN);
}

void loop() {
  if (Serial9bit.available()) {
    byte data = Serial9bit.read();
    if (seatalk.parse(data)) {
      // Seatalk message received
      switch (seatalk.getMessageId()) {
        case SEATALK_DEPTH:
          // Handle depth message
          float depth = seatalk.getDepth();
          break;
        // Handle other Seatalk message types here
      }
    }
  }
  
  // Send Seatalk message example
  uint8_t msg[5] = {SEATALK_DEPTH, 0x02, 0x3F, 0x00, 0x00};
  Serial9bit.write(msg, 5);
  delay(1000);
}

```

wellenvogel, 2023-03-16:
As far as I know the ESP32 hardware uart cannot really handle 9 Bits.
When going for bitbanging (Software serial) I'm not sure if this will work reliable in the multi threaded environment we use here.
What I found an interesting approach is to use rmt:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/rmt.html
https://github.com/naffej/esp32-rmt-uart
Would give all the flexibility we need for 9 Bit (and would give us additional uarts for lore nmea0183)
But finally it's a lot about testing. For me it's difficult as I do not have any seatalk device to work with.

hwater, 2023-03-16:
This looks really impressive. Did not know this, mabe i will time for an deeper Look in the next Lockdown.

Thanks a lot for your work in this boating community.

oxbown, 2024-03-22:
One idea...
ESP32 boards come with a protocol called RMT. The main purpose of the RMT is to transmit/receive IR signals
https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/peripherals/rmt.html

But in the magic world of programming..... some projects use the RMT to make custom protocols of transmission (like the PIO of the RP2040 of RPI pico)
https://github.com/naffej/esp32-rmt-uart
In that project use RMT to make custom UARTs
Could be adapted to read/write Seatalk.

wellenvogel, 2024-03-22:
yes - see my comment above.
I would consider this a really interesting feature - but I do not have any seatalk1 device for testing....

norbert-walter, 2024-03-22:
Hi Andreas, I have a old depth transducer ST60 Tridata from Raymarine with Seatalk1 protocol. If you want I can send you the transducer as test device.

oxbown, 2024-04-15:
[This guy](https://github.com/nauti-control/STWirelessRemote/) made a port of seatalk autopilot remote for esp32 board, the project use the [espsoftwareserial](https://github.com/plerup/espsoftwareserial) library for 9 bit uart coms

wellenvogel, 2024-04-15:
Yes, looks interesting. But without a device it will be hard...
In the moment I'm busy with some other tasks any way - but would be an interesting feature.