wellenvogel/esp32-nmea2000 #34: AP password not settable

W-Geronius, 2022-03-10:
Whatever I try, I cannot change the AP passwort (I have been trying on OBP60). 
Modifying Admin Pass + enabling/disabling seem to work.

Attached Screenshot and debug-log  was produced after factory reset. Changes to config as shown (testpass, testadmin) 
[image]

Debug log:
[2022-03-10_111325  COM8 ESP32 wroom.zip](https://github.com/wellenvogel/esp32-nmea2000/files/8222515/2022-03-10_111325.COM8.ESP32.wroom.zip)

After restart, the network name was changed. Logging in however does not work with the new password. It had remained the default.
Subsequent changes have no effect either.

wellenvogel, 2022-03-10:
Currently it works as designed (more or less).
As long as there is no hardware reset button (the define GWBUTTON_PIN must be set) the changing of the AP pass will not be allowed.
And the OBP60 does not set it currently.
But it should also not be visible in the UI in this case.
Additionally a flag FORCE_AP_PWCHANGE will be introduced to allow other software (like the OBP60) to force the ability of an AP password change (with the risk for the user to lock itself out and requiring a full reflash).

W-Geronius, 2022-03-11:
setting apPassword works fine in dev 20220311. Issue may be closed from my POV, thanks

closed 2022-03-14 by wellenvogel