wellenvogel/esp32-nmea2000 #35: dev20220311: please send <cr><lf> at line endings of log

W-Geronius, 2022-03-11:
not really an issue, but a sincere wish to make windows people happy .... it's hard to read like this:

[image]

wellenvogel, 2022-03-14:
I guess putty can do this easily:
https://the.earth.li/~sgtatham/putty/0.76/htmldoc/Chapter4.html#config-crlf

W-Geronius, 2022-03-14:
true, unfortunately putty is the only program offering this option - are there any drawbacks with <cr><lf> ?

W-Geronius, 2022-03-31:
thanks for trying to fix this - hmmm ... now however there is an excessive 0x0A:

[image]

wellenvogel, 2022-03-31:
nothing fixed until now.
Empty lines had been there already.
The "0d" are coming from the NMEA lines.

wellenvogel, 2022-03-31:
Should have been there all the time...