wellenvogel/esp32-nmea2000 #38: Valid GPS data by invalid NMEA0183 GPS sentences

norbert-walter, 2022-03-23:
Data shows valid GPS data by invalid NMEA0183 GPS sentences.

[image]

$GPRMC,000000.00,A,0.000,N,0.000,E,0.0,0.0,050505,0.0,E*6D
$GPXDR,U,12.340,V,Voltage*02
$GPRMC,000000.00,A,0.000,N,0.000,E,0.0,0.0,050505,0.0,E*6D
$GPRMC,,V,,,,,,,,,,N*53
$GPVTG,,,,,,,,,N*30
$GPGGA,,,,,,0,00,99.99,,,,,,*48
$GPGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99*30
$GPGSV,2,1,06,05,,,11,07,,,19,08,,,22,09,,,23*76
$GPGSV,2,2,06,15,,,24,21,,,21*7D
$GPGLL,,,,,,V,N*64
$GPRMC,000000.00,A,0.000,N,0.000,E,0.0,0.0,050505,0.0,E*6D
$GPXDR,U,12.330,V,Voltage*05
$GPRMC,000000.00,A,0.000,N,0.000,E,0.0,0.0,050505,0.0,E*6D
$GPRMC,,V,,,,,,,,,,N*53
$GPVTG,,,,,,,,,N*30
$GPGGA,,,,,,0,00,99.99,,,,,,*48
$GPGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99*30
$GPGSV,3,1,12,01,,,23,02,,,21,04,,,24,05,,,12*7D
$GPGSV,3,2,12,06,,,24,08,,,21,09,,,27,10,,,26*79
$GPGSV,3,3,12,11,,,25,14,,,24,15,,,21,21,,,21*79
$GPGLL,,,,,,V,N*64
$GPRMC,000000.00,A,0.000,N,0.000,E,0.0,0.0,050505,0.0,E*6D
$GPXDR,U,12.210,V,Voltage*06
$GPRMC,,V,,,,,,,,,,N*53
$GPVTG,,,,,,,,,N*30
$GPGGA,,,,,,0,00,99.99,,,,,,*48
$GPGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99*30
$GPRMC,000000.00,A,0.000,N,0.000,E,0.0,0.0,050505,0.0,E*6D
$GPGSV,3,1,10,01,,,22,04,,,23,05,,,26,06,,,21*7A

wellenvogel, 2022-03-23:
$GPRMC,000000.00,A,0.000,N,0.000,E,0.0,0.0,050505,0.0,E6D is a valid sentence.
But most probably it is generated by the gateway itself.
So maybe this should also be prevented...
Anyway:
Currently the used NMEA0183 lib has no great handling of empty fields. It simply parses them to 0.
For the special case of RMC we could at least use the status and ignore messages with status != A.

wellenvogel, 2022-03-23:
BTW: where did you get the log from - nearly none of the messages are valid NMEA messages.

norbert-walter, 2022-03-23:
All NMEA0183 sentences coming via serial interface from the NEO-6M by missing GPS fix. The sentences are the original NMEA0183 from UBlox. I'm not sure that the UBlox modul is a original modul. I remember me that other GPS USB sticks has the same empty values by missing GPS fix. I'm wondering about the poor date value 2041!

norbert-walter, 2022-03-23:
The checksum is ok. Leading zeros are suppressed. See voltage sentences.

wellenvogel, 2022-03-23:
This one is not ok:
$GPRMC,000000.00,A,0.000,N,0.000,E,0.0,0.0,050505,0.0,E6D
should be:
$GPRMC,000000.00,A,0.000,N,0.000,E,0.0,0.0,050505,0.0,E*6D
the "*" is missing.

norbert-walter, 2022-03-23:
Oh... right.

$GPXDR,U,12.330,V,Voltage*05
$GPRMC,110117.00,A,5111.65500,N,00647.78982,E,0.783,,230322,,,A*72
$GPVTG,,T,,M,0.783,N,1.450,K,A*2F
$GPGGA,110117.00,5111.65500,N,00647.78982,E,1,04,3.84,-16.1,M,46.5,M,,*47
$GPRMC,110117.00,A,5111.655,N,647.790,E,0.8,0.0,230322,0.0,E*5C
$GPGSA,A,3,19,02,24,12,,,,,,,,,4.58,3.84,2.50*0C
$GPGSV,3,1,12,02,30,103,26,03,03,354,,06,24,054,11,11,24,102,23*78
$GPGSV,3,2,12,12,67,067,26,19,07,039,14,22,34,298,16,24,33,143,20*7D
$GPGSV,3,3,12,25,67,274,,29,29,199,,31,17,305,15,32,39,273,20*75
$GPGLL,5111.65500,N,00647.78982,E,110117.00,A,A*65
$GPXDR,U,12.290,V,Voltage*0E
$GPRMC,110117.00,A,5111.655,N,647.790,E,0.8,0.0,230322,0.0,E*5C
$GPRMC,110118.00,V,,,,,,,230322,,,N*77
$GPVTG,,,,,,,,,N*30
$GPGGA,110118.00,,,,,0,04,3.53,,,,,,*5F
$GPGSA,A,1,19,02,24,12,,,,,,,,,3.67,3.53,1.00*09
$GPGSV,3,1,12,02,30,103,26,03,03,353,,06,24,054,12,11,24,102,23*7C
$GPGSV,3,2,12,12,67,067,26,19,07,039,13,22,34,298,17,24,33,143,21*7A
$GPRMC,110118.00,A,0.000,N,0.000,E,0.0,0.0,230322,0.0,E*62
$GPGSV,3,3,12,25,67,274,,29,29,199,,31,17,305,15,32,39,273,20*75
$GPGLL,,,,,110118.00,V,N*42
$GPXDR,U,12.280,V,Voltage*0F
$GPRMC,110118.00,A,0.000,N,0.000,E,0.0,0.0,230322,0.0,E*62
$GPRMC,110119.00,V,,,,,,,230322,,,N*76
$GPVTG,,,,,,,,,N*30

Now coming all sentences with correct check sums. Hmm...????

norbert-walter, 2022-03-23:
Ahh... the inline editor has a problem with the character.

`$GPXDR,U,12.330,V,Voltage*05
$GPRMC,110117.00,A,5111.65500,N,00647.78982,E,0.783,,230322,,,A*72
$GPVTG,,T,,M,0.783,N,1.450,K,A*2F
$GPGGA,110117.00,5111.65500,N,00647.78982,E,1,04,3.84,-16.1,M,46.5,M,,*47
$GPRMC,110117.00,A,5111.655,N,647.790,E,0.8,0.0,230322,0.0,E*5C
$GPGSA,A,3,19,02,24,12,,,,,,,,,4.58,3.84,2.50*0C
$GPGSV,3,1,12,02,30,103,26,03,03,354,,06,24,054,11,11,24,102,23*78
$GPGSV,3,2,12,12,67,067,26,19,07,039,14,22,34,298,16,24,33,143,20*7D
$GPGSV,3,3,12,25,67,274,,29,29,199,,31,17,305,15,32,39,273,20*75
$GPGLL,5111.65500,N,00647.78982,E,110117.00,A,A*65
$GPXDR,U,12.290,V,Voltage*0E
$GPRMC,110117.00,A,5111.655,N,647.790,E,0.8,0.0,230322,0.0,E*5C
$GPRMC,110118.00,V,,,,,,,230322,,,N*77
$GPVTG,,,,,,,,,N*30
$GPGGA,110118.00,,,,,0,04,3.53,,,,,,*5F
$GPGSA,A,1,19,02,24,12,,,,,,,,,3.67,3.53,1.00*09
$GPGSV,3,1,12,02,30,103,26,03,03,353,,06,24,054,12,11,24,102,23*7C
$GPGSV,3,2,12,12,67,067,26,19,07,039,13,22,34,298,17,24,33,143,21*7A
$GPRMC,110118.00,A,0.000,N,0.000,E,0.0,0.0,230322,0.0,E*62
$GPGSV,3,3,12,25,67,274,,29,29,199,,31,17,305,15,32,39,273,20*75
$GPGLL,,,,,110118.00,V,N*42
$GPXDR,U,12.280,V,Voltage*0F
$GPRMC,110118.00,A,0.000,N,0.000,E,0.0,0.0,230322,0.0,E*62
$GPRMC,110119.00,V,,,,,,,230322,,,N*76
$GPVTG,,,,,,,,,N*30
`

wellenvogel, 2022-04-12:
20220403

closed 2022-04-12 by wellenvogel