wellenvogel/esp32-nmea2000 #41: v20220417dev: 'tcp client connection closed on x.x.x.x' displays misleading IP adress

W-Geronius, 2022-04-17:
test setup: 
ATOM (10.10.10.144), as client, listening to TCP server (OP) at 10.10.10.1:10110
PC (10.10.10.81) as client, listening to TCP server (ATOM) at 10.10.10.144:10110

When PC stops listening, (sometimes) an error message is output on ATOM USB: 'tcp client connection closed on 10.10.10.1'
The connection ATOM-OP however is not interrupted - the message should read 'tcp client connection closed on **10.10.10.81**'

Not a biggie, but might be misleading when hunting for errors

wellenvogel, 2022-04-18:
This message is really only issued from the tcp client code on the ESP32. So it is not related at all to the server connections (i.e. the ones towards your PC).
Close messages for the server part are only visible with debug enabled (as they are not really treated as an error).
How did you check if the connection between the atom and OP remains intact?

W-Geronius, 2022-04-19:
As I noted: no biggie at all. 
Just should be kept in mind (documented) for bughunters. No need to dig into that.
Atom<-> OP connection, as well as ATOM tcp out was monitored and logged elsewhere while I was inspecting via PC.

wellenvogel, 2024-03-21:
No activity....

closed 2024-03-21 by wellenvogel