wellenvogel/esp32-nmea2000 #43: Provide an AWA correction angle

pljakobs, 2022-05-21:
I use this firmware to connect my Nasamarine masthead unit to nmea2000 without using an actual NASA wind display. 
The wind angle the unit provides is 0-359° relative to some reference forward direction that may or may not correspond with the actual forward direction of the boat, depending on masthead mounting and other factors.
I have patched in a fixed correction angle but it would be great to provide a settable correction through the UI.
It might even be possible to implement an auto calibration mode that watches a number of tacks and configures the wind correction angle to half the average tack angle.

wellenvogel, 2022-05-21:
Basically I agree that this would be a nice feature.
But currently the whole unit is designed as a thumb converter.
Finally you most probably would like to have similar functions for a lot of other values too.
So this would require a more generic concept.
Maybe if there is more interest on such functionality I could think about....

pljakobs, 2022-05-23:
hmm... would it really be such a huge change?
What I have in mind is: 
- a set of stored correction values
- a UI tab where the user can configure them
- the logic to apply them per converter

I could think of 
- AWA offset
- DPT keel offset
- RSA offset

I guess there are a few other values that this could be interesting for, but those are the ones that might differ due to construction. Maybe fluid levels could be interesting to adapt as well.

wellenvogel, 2022-05-23:
I guess it's a bit more complex.
Having some settings for the UI is straight forward.
But then:
For which conversion would you like to apply something?
N2K to 0183?
0183 to N2K?
0183 to 0183?
And additionally you would need to define the exact messages and values that this would apply to.
And maybe even some instance id's...
I agree that it would be simple for one exact use case.
But for a generic solution this is really a lot more complex.

pljakobs, 2022-05-24:
ok, I have been thinking of this as just a "attach one device to the n2k bus" solution in which case it's easy enough, but you're right.
I would generally look at n2k as the source of truth, as that will be the values that are available to other instruments and there would be no way to know if they use the same corrected instance or not.

wellenvogel, 2024-03-21:
as there is no further activity I  will close.

closed 2024-03-21 by wellenvogel

pljakobs, 2024-12-21:
building off the most recent code, I noticed that you added a few ui elements to configure wind angle conversions in general, I feel that this might have a place to live somewhere. I understand that the directionality is still worth thinking about, but it would really help my case (and I guess others, too) to be able to configure a correction angle.
Maybe a "wind source" selector and with it a correction?