wellenvogel/esp32-nmea2000 #44: Unit conversion problems in NMEA013 library

speters, 2022-06-30:
I discovered some unit conversion problems in https://github.com/ttlappalainen/NMEA0183/issues/37 .

Whileas it would be simple to fix this in 
https://github.com/wellenvogel/esp32-nmea2000/blob/78ed97f8a68d9feaa2b563d11bba9b252e401152/lib/nmea0183ton2k/NMEA0183DataToN2K.cpp#L884
and
https://github.com/wellenvogel/esp32-nmea2000/blob/78ed97f8a68d9feaa2b563d11bba9b252e401152/lib/nmea2kto0183/N2kDataToNMEA0183.cpp#L1049
I hope that @ttlappalainen will update his library upstream.

This issue can be seen in the line with https://github.com/wellenvogel/esp32-nmea2000/issues/43 where it seems to be favorable to have some means of control over filtering/conversion of the data being passed around, as I suspect there are devices in the wild which send rad/s, deg/s, rad/min, deg/min, ... without caring for units. Another possibility would be to calculate values from existing data (e.g. calculate ROT from heading change over time).

This project is so great and so much effort seems to have flown into all this complex code. It would be great to see this evolve from silver to gold.

wellenvogel, 2022-06-30:
Ok, will have a look at this.

wellenvogel, 2022-11-16:
Sorry for the long delay.
I' ve implemented the factor right now on my side. As I do not have any test data - could you try this change?
I would like to wait with a new release until we have some testing...
So If you need a prebuild firmware I can build one.

wellenvogel, 2022-11-17:
Do you maybe have some test data containing rot from real life?
I could create some artificial ones an test e.g. against canboat - but that is not the real truth...

ttlappalainen, 2022-11-17:
Have you tested my NMEA Simulator?

wellenvogel, 2022-11-17:
Hi Timo,
yes I was heavily using it for testing.
But at the end it's a simulator again .
I'm always happy to additionally have some "real world" data for testing.
Already had a couple of interesting experiences on what devices send...

wellenvogel, 2023-03-19:
https://github.com/wellenvogel/esp32-nmea2000/releases/tag/20230309

closed 2023-03-19 by wellenvogel