wellenvogel/esp32-nmea2000 #50: GPS/AIS to 2k (SNR way to high)

hwater, 2023-03-19:
Hello, are you shure the Values are correct. Here SNR is way to high. Data in serial from AIS/GPS. Results in SignalK.

{
  "count": 17,
  "satellites": [
    {
      "id": 10,
      "elevation": 0.4712,
      "azimuth": 0.1745,
      "SNR": 162
    },
    {
      "id": 44,
      "elevation": 0.5585,
      "azimuth": 0.7156,
      "SNR": 65
    },
    {
      "id": 48,
      "elevation": 0.2967,
      "azimuth": 0.2618,
      "SNR": 319
    },
    {
      "id": 42,
      "elevation": 0.1745,
      "azimuth": 0.2618,
      "SNR": 59
    },
    {
      "id": 45,
      "elevation": 0.1396,
      "azimuth": 0.6632,
      "SNR": 181
    },
    {
      "id": 46,
      "elevation": 0.2967,
      "azimuth": 0.2618,
      "SNR": 319
    },
    {
      "id": 34,
      "elevation": 0.2967,
      "azimuth": 0.2269,
      "SNR": 319
    },
    {
      "id": 33,
      "elevation": 0.2967,
      "azimuth": 0.2269,
      "SNR": 319
    },
    {
      "id": 41,
      "elevation": 0.0175,
      "azimuth": 1.0647,
      "SNR": 303
    },
    {
      "id": 40,
      "elevation": 0.0175,
      "azimuth": 1.0821,
      "SNR": 304
    },
    {
      "id": 36,
      "elevation": 0.2967,
      "azimuth": 0.2269,
      "SNR": 319
    },
    {
      "id": 51,
      "elevation": 0.1745,
      "azimuth": 0.2443,
      "SNR": 60
    },
    {
      "id": 39,
      "elevation": 0.0175,
      "azimuth": 1.0821,
      "SNR": 304
    },
    {
      "id": 47,
      "elevation": 0.2443,
      "azimuth": 0.2793,
      "SNR": 288
    },
    {
      "id": 43,
      "elevation": 0.5585,
      "azimuth": 0.7156,
      "SNR": 65
    },
    {
      "id": 50,
      "elevation": 0.1745,
      "azimuth": 0.2443,
      "SNR": 60
    },
    {
      "id": 0,
      "elevation": 0,
      "azimuth": 0,
      "SNR": 0
    }
  ]
}

this is normal GPS Data from an Moitessier HAT/serial:

{
  "count": 8,
  "satellites": [
    {
      "id": 85,
      "elevation": 1.1519173065792743,
      "azimuth": 5.759586532896371,
      "SNR": 35
    },
    {
      "id": 70,
      "elevation": 1.012290966387847,
      "azimuth": 3.4732052122617514,
      "SNR": 0
    },
    {
      "id": 84,
      "elevation": 0.7853981635767779,
      "azimuth": 2.146754980443193,
      "SNR": 0
    },
    {
      "id": 86,
      "elevation": 0.349065850478568,
      "azimuth": 5.497787145037446,
      "SNR": 29
    },
    {
      "id": 71,
      "elevation": 0.15707963271535558,
      "azimuth": 3.6302848449771066,
      "SNR": 0
    },
    {
      "id": 77,
      "elevation": 0.15707963271535558,
      "azimuth": 6.091199090851011,
      "SNR": 0
    },
    {
      "id": 76,
      "elevation": 0.10471975514357039,
      "azimuth": 5.305800927274233,
      "SNR": 20
    },
    {
      "id": 68,
      "elevation": 0.0698131700957136,
      "azimuth": 0.5759586532896371,
      "SNR": 0
    }
  ]
}

hwater, 2023-03-21:
now i am back to an usb direct connection, without the converter. all seems to be back to normal.
it must be an qoirk with the NMEA0183 to Can converter. so it lies in the Library. i will ask there. sorry for the traffic.
many greetings - holger

{
  "count": 11,
  "satellites": [
    {
      "id": 3,
      "elevation": 1.3788101093903435,
      "azimuth": 0.261799387858926,
      "SNR": 45
    },
    {
      "id": 4,
      "elevation": 0.9250245037682051,
      "azimuth": 3.5255650898335364,
      "SNR": 29
    },
    {
      "id": 1,
      "elevation": 0.8726646261964199,
      "azimuth": 2.548180708493546,
      "SNR": 30
    },
    {
      "id": 17,
      "elevation": 0.6632251159092791,
      "azimuth": 4.590215933793169,
      "SNR": 47
    },
    {
      "id": 19,
      "elevation": 0.5585053607657088,
      "azimuth": 5.1836278796067345,
      "SNR": 49
    },
    {
      "id": 21,
      "elevation": 0.5061454831939235,
      "azimuth": 2.583087293541403,
      "SNR": 0
    },
    {
      "id": 31,
      "elevation": 0.48869219066999514,
      "azimuth": 1.1693705991032026,
      "SNR": 42
    },
    {
      "id": 28,
      "elevation": 0.40142572805035315,
      "azimuth": 0.8901179187203483,
      "SNR": 35
    },
    {
      "id": 9,
      "elevation": 0.349065850478568,
      "azimuth": 3.8048177702163906,
      "SNR": 17
    },
    {
      "id": 6,
      "elevation": 0.29670597290678274,
      "azimuth": 5.4279739749417315,
      "SNR": 44
    },
    {
      "id": 12,
      "elevation": 0,
      "azimuth": 6.108652383374939,
      "SNR": 0
    }
  ]
}

wellenvogel, 2023-03-25:
Thanks for bringing this up.
Do you have some test data I can use to reproduce the issue?
If I understand it correctly the problem occurs when translating from 0183 to 2K?

hwater, 2023-03-25:
Yes, it translates most gps/ais data okay, also the depth from the serial to can, but the snr is too high. Now directly from 0183 to SK all the data is correct (without the Gateway).Ich werde versuchen es nochmal anzuschließen um dir Daten aufzunehmen. Am besten parallel can Output mit Analyzer und 0183 Input. Liebe Grüße Andreas von der donau.Holger - typed mobil.https://donautics.stwst.at/stories-donautics - typed mobil.https://donautics.stwst.at/stories-donautics

wellenvogel, 2024-03-21:
any chance to get some test data?

hwater, 2024-03-21:
Hello Andreas, sorry for the long delay, i had some healthy issures last year.

Today i was life on board. I connected the ESP2NMEA in parallel to the EasyAIS2 38400 B. (only RXD)
The data comes in and goes out via N2K-Bus if this is enabled. But i cant´t acces the USB Bus from remote.
But i could get it via UDP or TCP from the pi.

On SignalK the EasyAIS is normaly connected via serial Port in. This works nearly an Year now correctly.

Tested again:
The AIS/GPS Data so goes serial to SK and also via N2K to the SK and shows the false SNR Values here like last Time.
The ESP software is upped to the last version from 2024/12/23

What is the best way in config.json to generate the DATA you need? 
I have remote Accesss to the pi on Ship, so i can upload any version you send me also to the ESP.

Holger

if you prefer, we also can talk in german. 

Holger

wellenvogel, 2024-03-22:
Hallo Holger,
erst mal eine Frage zum Verständnis:
Du hast GPS Daten auf 0183, nimmst den Konverter um N2K zu erzeugen und liest das dann auf dem Pi ein (Signalk)?

Dann wäre es gut mal die 0183 Daten zu haben (kannst Du z.B. vom Konverter per TCP holen). Dann kann ich ja selbst mal schauen, was bei mir daraus wird.
Oder Du holst sie von SignalK - sollte ja auch gehen (port 10110).

wellenvogel, 2024-11-14:
https://github.com/wellenvogel/esp32-nmea2000/releases/tag/20241114

closed 2024-11-14 by wellenvogel