wellenvogel/esp32-nmea2000 #52: Add Support to M5AtomS3

oxbown, 2023-06-13:
Please consider to include [AtomS3](https://docs.m5stack.com/en/core/AtomS3%20Lite) to the build list, with[ platformio ](https://docs.platformio.org/en/latest/boards/espressif32/m5stack-atoms3.html?utm_source=platformio&utm_medium=piohome)this could be really easy.

free-x, 2023-06-21:
I compared board definitions of ATOM and ATOM-S3. It strikes me that S3 has no "CAN" connectivity.
Are you sure that S3 is right plattform?

oxbown, 2023-06-22:
Espressif renames CAN BUS to [TWAI](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/peripherals/twai.html) so with new boards libraries like "NMEA2000_esp32" not compile.

References to this bug [here](https://github.com/ttlappalainen/NMEA2000_esp32/issues/14)

A possible solution [here](https://github.com/jiauka/NMEA2000_esp32xx) --> Commented [here](https://github.com/ttlappalainen/NMEA2000_esp32/issues/17)

oxbown, 2023-09-05:
I'm trying to test the new twai version but something is not working properly. When I connect my m5atoms3 lite with the can bus unit to the N2K network seems that the atom can not read the nmea2k data.
The "NMEA2000 State" in the "Status" page witches between "[32] BUS_OFF" and "[32] OFFLINE", everything is connected correctly (including 120ohm resistances).
There is only one N2K device on the network (plus M5atoms3), an Airmar P79S deep transducer.

wellenvogel, 2023-09-05:
How did you build & install the firmware?
Did this set up ever work before?
How exactly did you connect the device? CanH/L + ground? Are you sure about the wires - sometimes colors are different!
And please enable the debug log via the UI, connect an terminal via USB and send the output (especially from start up).

free-x, 2023-09-06:
i cannot confirm the behavior. this works as expected
```
 $ pio run -e m5stack-atoms3-canunit
...
 $ pio run -e m5stack-atoms3-canunit -t upload
...
  $ pio run -e m5stack-atoms3-canunit -t monitor
$GPRMC,114135.23,A,5423.940,N,1111.837,E,3.3,196.3,170722,3.2,E*66
$GPMWV,49.8,R,9.2,M,A*17
$GPMWV,58.8,T,8.2,M,A*10
$GPMWD,58.8,T,55.6,M,16.0,N,8.2,M*6D
$GPHDG,197.4,,,3.2,E*3F
$GPHDT,200.6,T*31
$GPRSA,7.8,A,0.0,A*58
$GPVTG,195.8,T,192.6,M,3.3,N,6.1,K*40
$GPHDG,197.4,,,3.2,E*3F
$GPHDT,200.6,T*31
$GPRSA,7.4,A,0.0,A*54
$GPMWV,49.8,R,9.2,M,A*17
$GPMWV,58.9,T,8.2,M,A*11
$GPMWD,58.9,T,55.7,M,16.0,N,8.2,M*6D
....
```
I'm on master branch of upstream
```
$ git branch -v
* master 17615ad cleanup unused code, make fallback_serial working again
  twai   e61cfac example to lock idf version

~/git/esp32-nmea2000_twai$ git remote -v
origin	https://github.com/wellenvogel/esp32-nmea2000 (fetch)
origin	https://github.com/wellenvogel/esp32-nmea2000 (push)
```
greetings
free-x

oxbown, 2023-09-09:
Well...sorry, CAN works ok... i´d made a mistake wiring the devices... i can confirm that latest main works on ESP32-S3.

closed 2023-09-10 by wellenvogel