wellenvogel/esp32-nmea2000 #58: NMEA2000-Gateway-MultiHUB

oxbown, 2023-10-27:
Hi!
I've got an idea to expand the possibilities of NMEA2000-Gateway. M5Stack has a product to expand ATOM ports "[ATOMIC PortABC Extension Base](https://shop.m5stack.com/products/atomic-portabc-extension-base?variant=43894734979329)"
With that module, we could connect several devices to the atom or atoms3 (compatible with both boards). CAN bus, RS485, Serial TTL, and i2C could be connected at the same time.
This is an example of the idea:
[image]

wellenvogel, 2023-10-27:
Looks like a cool piece.
With the new approach of customized builds this could be integrated.
But we need someone to test this...

oxbown, 2023-10-27:
I´m going to buy the Hub when it is available in [Mouser Electronics](https://www.mouser.es/ProductDetail/170-A130)
My plan is to make a GATEWAY-SUPER-BOX-IP67 🤣  with all of that hardware inside.

oxbown, 2023-12-15:
Working on the idea...
[image]

wellenvogel, 2024-03-24:
https://github.com/wellenvogel/esp32-nmea2000/releases/tag/20240324

closed 2024-03-24 by wellenvogel