wellenvogel/esp32-nmea2000 #59: 9-DoF IMU for Magnetic heading and boat heeling.

oxbown, 2023-11-02:
I´m planning to use my [ICM-20948](https://www.mouser.es/ProductDetail/485-4554) to measure Magnetic heading and boat heeling. The sensor is i2c compatible, I use a [cable](https://www.mouser.es/ProductDetail/485-4528) to connect it to the M5atom groove port.

@wellenvogel do you think that is possible to add the sensor to the "GwIicTask"?? 

Library info: https://github.com/adafruit/Adafruit_ICM20X

wellenvogel, 2023-11-02:
Should not be that much of an issue...
But I have no device for testing.
And someone would need to write some minimal documentation.
If you only need this for your own you can easily write an own user task - just following the pattern from GwIicTask.

oxbown, 2023-11-03:
New product of M5stack [M5Unit-IMU-Pro](https://github.com/m5stack/M5Unit-IMU-Pro-Mini), could be a good candidate.

oxbown, 2023-11-03:
> Should not be that much of an issue... But I have no device for testing. And someone would need to write some minimal documentation. If you only need this for your own you can easily write an own user task - just following the pattern from GwIicTask.

My plan is to help add more capabilities to the project (writing some code and testing with my equipment)

wellenvogel, 2023-11-03:
That sounds great...
I just heavily restructured the IICSensorTask.
Now it should be much easier to write some code for a new sensor.
Basically you need to implement a Sensor config that inherits from SensorBase - see e.g. https://github.com/wellenvogel/esp32-nmea2000/blob/28ae7d3bbdf9072afd67e8b2e6c80970d558260b/lib/iictask/GwQMP6988.cpp#L24
and a register function that we include in the sensor task.
Additionally you need to add the necessary config to the [config.json](https://github.com/wellenvogel/esp32-nmea2000/blob/master/lib/iictask/config.json).
I prepared the sensors now for 2 buses and each 2 sensor instances on one bus. You can control the sensors you would like to have via build flags (and the automated build for that is also on the way...).
So if you are interested - maybe you would like to implement your sensor this way.

wellenvogel, 2024-03-21:
Are you still interested ?

oxbown, 2024-03-22:
Yes, I´m currently using an MPU9250 but not integrated into the ESP32 gateway. I use another esp32 to transmit the data (heading, pitch, roll) via TCP/IP in NMEA format.

wellenvogel, 2024-03-22:
Ok, sounds intersting. Do you have a github repo for that part?
At the end it's again about testing...

oxbown, 2024-03-26:
My test MPU board. I use a M5 proto to make the wiring.
[image]

wellenvogel, 2024-03-26:
cool.
In the moment I'm just busy with a couple of other things...
Not sure if you would like to go for a PR?

wellenvogel, 2024-11-13:
Are you still working on this?
What about a PR?

wellenvogel, 2025-03-04:
No further activity - so closing this.

closed 2025-03-04 by wellenvogel