wellenvogel/esp32-nmea2000 #61: I2c sensors ERRORs

oxbown, 2023-12-15:
Hello!
I´m working in a PortABC version of the esp32-nmea2000 code. I´m using pin 39 for SCL and pin 38 for SDA on a M5ATOMS3.
I´m traying to use a ENV III sensor using the "-D GWSHT3X11" and "-D GWQMP698811" flags on "platformio.ini".
Compilation is OK and upload to board is OK, but if i plug the sensor in the i2c port the board start to reboot endlessly.

This is the serial log:
`ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x2b (SPI_FAST_FLASH_BOOT)
Saved PC:0x40377870
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
entry 0x403c98d4
E (228) esp_core_dump_flash: No core dump partition found!
E (228) esp_core_dump_flash: No core dump partition found!
[   245][E][Preferences.cpp:483] getString(): nvs_get_str len fail: systemName NOT_FOUND
[   246][E][Preferences.cpp:483] getString(): nvs_get_str len fail: talkerId NOT_FOUND
[   250][E][Preferences.cpp:483] getString(): nvs_get_str len fail: stopApTime NOT_FOUND
[   258][E][Preferences.cpp:483] getString(): nvs_get_str len fail: apPassword NOT_FOUND
[   265][E][Preferences.cpp:483] getString(): nvs_get_str len fail: apIp NOT_FOUND
[   273][E][Preferences.cpp:483] getString(): nvs_get_str len fail: apMask NOT_FOUND
[   280][E][Preferences.cpp:483] getString(): nvs_get_str len fail: useAdminPass NOT_FOUND
[   288][E][Preferences.cpp:483] getString(): nvs_get_str len fail: adminPassword NOT_FOUND
[   296][E][Preferences.cpp:483] getString(): nvs_get_str len fail: showInvalidData NOT_FOUND
[   305][E][Preferences.cpp:483] getString(): nvs_get_str len fail: minXdrInterval NOT_FOUND
[   313][E][Preferences.cpp:483] getString(): nvs_get_str len fail: min2KInterval NOT_FOUND
[   321][E][Preferences.cpp:483] getString(): nvs_get_str len fail: sendN2k NOT_FOUND
[   328][E][Preferences.cpp:483] getString(): nvs_get_str len fail: usbActisense NOT_FOUND
[   336][E][Preferences.cpp:483] getString(): nvs_get_str len fail: usbBaud NOT_FOUND
[   344][E][Preferences.cpp:483] getString(): nvs_get_str len fail: sendUsb NOT_FOUND
[   351][E][Preferences.cpp:483] getString(): nvs_get_str len fail: receiveUsb NOT_FOUND
[   359][E][Preferences.cpp:483] getString(): nvs_get_str len fail: usbToN2k NOT_FOUND
[   367][E][Preferences.cpp:483] getString(): nvs_get_str len fail: usbReadFilter NOT_FOUND
[   375][E][Preferences.cpp:483] getString(): nvs_get_str len fail: usbWriteFilter NOT_FOUND
[   383][E][Preferences.cpp:483] getString(): nvs_get_str len fail: usbActSend NOT_FOUND
[   391][E][Preferences.cpp:483] getString(): nvs_get_str len fail: serialDirection NOT_FOUND
[   400][E][Preferences.cpp:483] getString(): nvs_get_str len fail: sendSerial NOT_FOUND
[   407][E][Preferences.cpp:483] getString(): nvs_get_str len fail: receiveSerial NOT_FOUND
[   415][E][Preferences.cpp:483] getString(): nvs_get_str len fail: serialToN2k NOT_FOUND
[   423][E][Preferences.cpp:483] getString(): nvs_get_str len fail: serialReadF NOT_FOUND
[   431][E][Preferences.cpp:483] getString(): nvs_get_str len fail: serialWriteF NOT_FOUND
[   439][E][Preferences.cpp:483] getString(): nvs_get_str len fail: serial2Dir NOT_FOUND
[   447][E][Preferences.cpp:483] getString(): nvs_get_str len fail: sendSerial2 NOT_FOUND
[   454][E][Preferences.cpp:483] getString(): nvs_get_str len fail: receiveSerial2 NOT_FOUND
[   463][E][Preferences.cpp:483] getString(): nvs_get_str len fail: serial2ToN2k NOT_FOUND
[   471][E][Preferences.cpp:483] getString(): nvs_get_str len fail: serial2ReadF NOT_FOUND
[   479][E][Preferences.cpp:483] getString(): nvs_get_str len fail: serial2WriteF NOT_FOUND
[   487][E][Preferences.cpp:483] getString(): nvs_get_str len fail: serverPort NOT_FOUND
[   494][E][Preferences.cpp:483] getString(): nvs_get_str len fail: maxClients NOT_FOUND
[   502][E][Preferences.cpp:483] getString(): nvs_get_str len fail: sendTCP NOT_FOUND
[   510][E][Preferences.cpp:483] getString(): nvs_get_str len fail: readTCP NOT_FOUND
[   517][E][Preferences.cpp:483] getString(): nvs_get_str len fail: tcpToN2k NOT_FOUND
[   525][E][Preferences.cpp:483] getString(): nvs_get_str len fail: tcpReadFilter NOT_FOUND
[   533][E][Preferences.cpp:483] getString(): nvs_get_str len fail: tcpWriteFilter NOT_FOUND
[   541][E][Preferences.cpp:483] getString(): nvs_get_str len fail: sendSeasmart NOT_FOUND
[   549][E][Preferences.cpp:483] getString(): nvs_get_str len fail: tclEnabled NOT_FOUND
[   557][E][Preferences.cpp:483] getString(): nvs_get_str len fail: remotePort NOT_FOUND
[   565][E][Preferences.cpp:483] getString(): nvs_get_str len fail: remoteAddress NOT_FOUND
[   573][E][Preferences.cpp:483] getString(): nvs_get_str len fail: sendTCL NOT_FOUND
[   581][E][Preferences.cpp:483] getString(): nvs_get_str len fail: readTCL NOT_FOUND
[   588][E][Preferences.cpp:483] getString(): nvs_get_str len fail: tclToN2k NOT_FOUND
[   596][E][Preferences.cpp:483] getString(): nvs_get_str len fail: tclReadFilter NOT_FOUND
[   604][E][Preferences.cpp:483] getString(): nvs_get_str len fail: tclWriteFilter NOT_FOUND
[   612][E][Preferences.cpp:483] getString(): nvs_get_str len fail: tclSeasmart NOT_FOUND
[   620][E][Preferences.cpp:483] getString(): nvs_get_str len fail: wifiClient NOT_FOUND
[   628][E][Preferences.cpp:483] getString(): nvs_get_str len fail: wifiSSID NOT_FOUND
[   635][E][Preferences.cpp:483] getString(): nvs_get_str len fail: wifiPass NOT_FOUND
[... 129 lines cut ...]
[  1658][E][Preferences.cpp:483] getString(): nvs_get_str len fail: BME28022iid NOT_FOUND
[  1666][E][Preferences.cpp:483] getString(): nvs_get_str len fail: BME28022intv NOT_FOUND
[  1674][E][Preferences.cpp:483] getString(): nvs_get_str len fail: BME28022tmNam NOT_FOUND
[  1682][E][Preferences.cpp:483] getString(): nvs_get_str len fail: BME28022huNam NOT_FOUND
[  1690][E][Preferences.cpp:483] getString(): nvs_get_str len fail: BME28022prNam NOT_FOUND
LOG:1850:GWWIFI: AP auto shutdown disabled (0s)
GWSERIAL:1950:wifi AP pass: esp32nmea2k
GWSERIAL:1950:admin pass: esp32admin
GWSERIAL:1950:initialized IIC 1 at sda=38,scl=39
Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception). 
Debug exception reason: Stack canary watchpoint triggered (iicTask) 
Core  0 register dump:
PC      : 0x40380ce7  PS      : 0x00060836  A0      : 0x80383944  A1      : 0x3fcc45a0  
A2      : 0x3fce9f40  A3      : 0xb33fffff  A4      : 0x0000abab  A5      : 0x00060823  
A6      : 0x00060820  A7      : 0x0000cdcd  A8      : 0xb33fffff  A9      : 0xffffffff  
A10     : 0x3fcc49a0  A11     : 0x00000000  A12     : 0x624dd2f2  A13     : 0x404e1810  
A14     : 0x02ce9f40  A15     : 0x00ffffff  SAR     : 0x0000000a  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xffffffff  

Backtrace: 0x40380ce4:0x3fcc45a0 0x40383941:0x3fcc45e0 0x40383bad:0x3fcc4600 0x403780a9:0x3fcc4620 0x403780bd:0x3fcc4650 0x403780ea:0x3fcc4670 0x40383f91:0x3fcc4690 0x420b8c5e:0x3fcc46b0 0x420b2d5a:0x3fcc4750 0x420b043d:0x3fcc4a60 0x42026f06:0x3fcc4b20 0x4202fceb:0x3fcc4b90 0x4202d292:0x3fcc4c20 0x4200662e:0x3fcc4c70 0x4200681e:0x3fcc4cd0 0x42023d6e:0x3fcc4cf0 0x420a858b:0x3fcc4d20 0x420a87ed:0x3fcc4e80 0x420c50f9:0x3fcc4ed0 0x420a74c5:0x3fcc4ef0 0x420c2d39:0x3fcc4fa0

ELF file SHA256: afac89f3bba3464c

E (3513) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...`

wellenvogel, 2023-12-15:
Not sure what you modified...
Stack canary watchpoint triggered means that the stack size is exceeded (you can easily google it). So this would mean that the iictask needs more stack (why?).
You could run the decoder.py from the tools dir to get a source stack.

oxbown, 2023-12-15:
This is what I change
https://github.com/oxbown/esp32-nmea2000/commit/3a366129d127334d9e19d24a58a25963171e70ea

oxbown, 2023-12-15:
Solved here!! https://github.com/oxbown/esp32-nmea2000/commit/378b4e269cae8c44e2993f3b666311b3418cee59

closed 2023-12-17 by oxbown