wellenvogel/esp32-nmea2000 #65: XDR Mappings PGN-127506 (DC Detailed Status)

oxbown, 2024-01-04:
I´m using an esp8266 to send the battery information over TCP to the esp32-nmea2K. Messages are in nmea0183 format with XDR sentences
For example:
`$JCXDR,U,12.0,V,I,2.5,A,C,25.0,C,BAT1*hh`

I can read voltage, amperage, and battery temperature on esp32 using XDR mappings with that sentence.

But I have more information about the battery, in particular, 2 more values that are interesting
State of Charge (SoC) - The remaining percentage of charge (%)
Time Till Zero Charge (TTZ) - Time to discharge the battery (seconds)

These two values can be included in the XDR sentence:
`$JCXDR,U,12.0,V,I,2.5,A,C,25.0,C,SOC,50.0,,TTZ,3520,,BAT1*hh`

The n2k PNG that has that information is 127506. That sentence is not present in the project, but fortunately, the nmea2000 library supports it!
https://github.com/ttlappalainen/NMEA2000/blob/5e4fb0a49f4832f134878054da1ce8e4d700ee55/src/N2kMessages.h#L1981

The question is. Can be integrated? is there any challenge?

wellenvogel, 2024-01-04:
Most probably it can be integrated.
Would require some idea about the XDR unit/type (any standard for that?) and the implementation of the appropriate definitions for the XDR mappings and the implementation in the converter.
Do you already use the n2k lib on the esp8266? 
If yes you could potentially directly send the data in n2k format using the seasmart format via TCP.
This would allow you to implement things independently...
Only drawback: No value display on the converter.

oxbown, 2024-01-04:
I did not find any standard for that XDR, the only thing that I found is XDR can be expanded or adapted by the manufacturer.
Format: $--XDR,a,x.x,b,.....,c *hh
**_--_** is the [Talkder id ](https://gpsd.gitlab.io/gpsd/NMEA.html#_talker_ids)
**_a_** is the Transducer Type
**_x.x_** is the Value of the measure
**_b_** is the Unit of the measure
**_c_** is the Name of the transducer
**_hh_** is the checksum

There is a list of standard Transducer Types:
`A - Angular displacement
	//  C - Temperature
	//  D - Depth
	//  F - Frequency
	//  H - Humidity
	//  N - Force
	//  P - Pressure
	//  R - Flow
	//  B - Absolute humidity
	//  G - Generic
	//  I - Current
	//  L - Salinity
	//  S - Switch, valve
	//  T - Tachometer
	//  U - Voltage
	//  V - Volume`

And a list of standard Units 
`A - Amperes
	//  B - Bars | Binary
	//  C - Celsius
	//  D - Degrees
	//  H - Hertz
	//  I - liters/second
	//  K - Kelvin or Density, kg/m3 kilogram per cubic meter
	//  M - Meters or Cubic Meters (m3)
	//  N - Newton
	//  P - percent of full range or Pascal
	//  R - RPM
	//  S - Parts per thousand
	//  V - Volts`

Transducer types can be customized (like: "MAGX", "SOC", "TTZ")
Units can also be custom o directly not units.

I´m using esp8266 with micropython so I can not use the library. The idea is to make sensors or devices with a few lines of code.

wellenvogel, 2024-01-04:
Yes I know about the types/units you mentioned. But none of them really fit's.
So we would end up with some proprietary mappings. Doable - but not really great.
And btw: Transducer types are one character, the names are free...

Basically to support any kind of user defined mapping we would need to extend the mapping definitions to also be able to set the transducer type, the unit and the PGN+field you would like to map to.

oxbown, 2024-01-04:
Here is a reference in other project https://github.com/maritime-labs/calypso-anemometer/issues/21#issuecomment-1444396266 for battery level.
Uses L for "level" and R for "ratio" unit