wellenvogel/esp32-nmea2000 #68: Generic RS-232/RS-422 unit (Full Duplex Serial)

oxbown, 2024-03-26:
There is one thing that can make better de CI online build service.
M5stack don't sell a RS232/422 unit, but there is a lot of boards available on the market.
[image]

wellenvogel, 2024-03-26:
Sounds good.
PR?

oxbown, 2024-03-26:
> Sounds good.
> PR?

I´ll try...
This hardware configuration is functional?  https://github.com/wellenvogel/esp32-nmea2000/blob/146bc44c6202c5b903835165c691030610ed3915/lib/hardware/GwM5Grove.in#L36

oxbown, 2024-03-30:
Tested latest [testci20240328](https://www.wellenvogel.net/software/esp32/cibuild.html?tag=testci20240328) and works well. Able to read and write NMEA0183 data with real hardware.
Baud rate has to be selected in config to 4800 to match with the nmea standard.
I´ll make a PR with docs.
Thanks!

wellenvogel, 2025-03-04:
Do you still work on this? I will close it now - feel free to reopen.

closed 2025-03-04 by wellenvogel