wellenvogel/esp32-nmea2000 #70: [Request] M5Stack Core controllers support?

pissten, 2024-03-29:
Hi,

Thanks for your awesome and great work!
Would you be interessted in adding support for the Core controllers as well?
Seems like a Core controller with a commu and gps module has all the interfaces 
needed to connect whatever we want, to this great NMEA gateway/multiplexer code
of yours. Right?
I was thinking of IMU and canbus NMEA 2K and 0183 with this setup

wellenvogel, 2024-03-30:
Good idea...
I personally do not have a core and connected devices.
So it's about providing the correct hardware defs, some documentation and - TESTING...

pissten, 2024-04-08:
I'll be more than willing to help testing, if you are up for the task on the software part ;)

wellenvogel, 2024-04-10:
Maybe I will give it a try. But will need some time - currently I'm busy with my other projects.