wellenvogel/esp32-nmea2000 #72: Add support for Lilygo T-Display-S3

ibravo, 2024-06-21:
I've been struggling for a couple of days now to make the Lilygo S3 work with the code. This device should be fairly similar to the configurations for the BOARD_HOMBERGER or the PLATFORM_BOARD_NODEMCU_32S. The board uses an external SN65HVD230 can transceiver module to connect to the NMEA network.

When using the same configuration as the BOARD_HOMBERGER, the compiler fails on needing a serial port, and when adding a   #define USBSerial Serial the board compiles but doesn't upload.

I'm making these changes on the GwHardware.h file

 Any guidance will be appreciated.

wellenvogel, 2024-06-21:
Would be good to see the code that you have modified...
But for an S3 you need to use a valid S3 config- different chipid.
Normally you should not modify the core software but go for some user extensions.
https://github.com/wellenvogel/esp32-nmea2000#extending-the-software
This way you can easily merge upstream changes.

ibravo, 2024-06-24:
I've modified the code in several places until I realized that the problem I was having about uploading was caused by not placing the board in boot mode. Some of the modifications might not be required, but didn't know enough about the code to know what was good to keep, and what not needed for the board to work.  Take a look at https://github.com/wellenvogel/esp32-nmea2000/commit/6a81253e27fc5b6099f1df2297bd3dc8e3eb5f7d for details and you might be able to point me into the correct direction.

Aside from this, once loaded, I had an issue with the password not being recognized on an iPhone, so updated the password to 'password' and that took me into the code. BTW, great program. I had the opportunity to test it this Sunday at the boat and the messages were reaching the little device.

My next objective will be to implement the display code to show messages on the board itself.

wellenvogel, 2024-06-24:
OK, so not sure how I can further assist...
Some hints from the commit:
(1) create a new directory for your board - like lilygotask (must have a task in the name) and put your stuff there
(2) do not change any code in the base outside your directory (you have a platformio.ini and your header , you can create a task for the display control and any config's you need in the UI).
This way you will always be able to merge in upcoming upstream changes.

unxs0, 2024-06-24:
This is the new espressif TWAI S3 problem. See https://github.com/ttlappalainen/NMEA2000_esp32/issues/25.

wellenvogel, 2024-06-24:
Sorry, I do not understand this.
My code already uses an own twai based implementation.
Is there anything that does not work with the lilygo?

ibravo, 2024-06-25:
> OK, so not sure how I can further assist... Some hints from the commit: (1) create a new directory for your board - like lilygotask (must have a task in the name) and put your stuff there (2) do not [...]

Thanks. I'll try to create this directory and put my stuff there to keep it in synch with the upstream code.
Will share again once created as it might help others on how to do it.

wellenvogel, 2024-06-25:
Sounds good.

wellenvogel, 2024-11-13:
Closing this due to no further activity.

closed 2024-11-13 by wellenvogel