wellenvogel/esp32-nmea2000 #73: LILYGO® TTGO T-CAN485 ESP32

oxbown, 2024-06-25:
[New board from TTGO](https://www.lilygo.cc/products/t-can485) could be a good choice for the project.
ESP32 + CAN (isolated) + RS485 (isolated) + 12vDC-->5vDC

[image]

All in one board suitable for marine communications!!!

wellenvogel, 2024-06-25:
Thanks a lot.
Really looking cool.

wellenvogel, 2025-03-04:
Have you already been able to test this board?
I would be slightly afraid on the power supply. Max rating is 16V and there is no real protection against spikes - so I'm not sure if this would be sufficient for normal operation on board (we already could have e.g. 14.8 V from the regulator - so there is not much headroom...).