wellenvogel/esp32-nmea2000 #75: True Wind Calculation

quantenschaum, 2024-08-20:
First, thanks for this project! :tophat: 

related to https://github.com/wellenvogel/avnav/issues/367 and https://github.com/kdschmidt1/Sail_Instrument

I noticed that the gateway computes true wind from COG/SOG and AWA/AWS, but I suspect the calculations not to be correct.

## Testing

I fed the gateway with the following test data (as return by the GW in `PCDIN`) and got this output. The test data values were generated using [`CourseData`](https://github.com/quantenschaum/Sail_Instrument/blob/1c464da0dfe77f230d7558e8f6d665e590159f30/Sail_Instrument/plugin.py#L653).

### 1. No Current

- SET/DFT = 0 and thus 
  - COG/SOG = HDT/STW and 
  - true wind = ground wind

```python
print(CourseData(SET=90,DFT=0,HDT=90,STW=5,AWA=-50,AWS=15,angles360=1))
```

```
AWA=-50
AWS=15
COG=90.0
DFT=0
GWD=21.99692847308269
GWS=12.392814795155259
HDT=90
SET=90
SOG=5.0
STW=5
TWA=291.9969284730827
TWD=21.996928473082676
TWS=12.39281479515526
```

GW output

```nmea
$PCDIN,01F802,0000EF27,14,FFFC5C3D0101FFFF*2A
$GPVTG,90.0,T,,M,5.0,N,9.3,K*56
$PCDIN,01FD02,0000EF28,14,FF040359D302FFFF*23
$GPMWV,310.0,R,7.7,M,A*2B
$GPMWV,292.0,T,6.4,M,A*24
$GPMWD,292.0,T,292.0,M,12.4,N,6.4,M*66
```

- `VTG` COG/SOG = 90°/5kn :heavy_check_mark: 
- `MWV-R` AWA/AWS =  310°/7.7m/s :heavy_check_mark: 
- `MWV-T` TWA/TWS = 292°/6.4m/s :heavy_check_mark: 
- `MWD` TWD/TWS 292°/12.4kn :x: same as `MWV-T` but should be [true wind *direction* not angle](https://gpsd.gitlab.io/gpsd/NMEA.html#_mwd_wind_direction_speed) (need to add HDT) Forwarding this to AvNav results in a wrong TWD.

### 2. With Current

- SET/DFT != 0 and thus 
  - COG/SOG != HDT/STW and 
  - true wind != ground wind

```python
print(CourseData(SET=30,DFT=2,HDT=90,STW=5,AWA=-50,AWS=15,angles360=1))
```

```
AWA=-50
AWS=15
COG=73.89788624801398
DFT=2
GWD=20.46505232502011
GWS=10.416016194605621
HDT=90
SET=30
SOG=6.244997998398398
STW=5
TWA=291.9969284730827
TWD=21.996928473082676
TWS=12.39281479515526
```

GW output

```nmea
$PCDIN,01F802,0019FB96,14,FFFC62324101FFFF*2F
$GPVTG,73.9,T,,M,6.2,N,11.6,K*6F
$PCDIN,01F503,0019FB98,14,FF0101FFFF00FFFF*29
$GPVHW,90.0,T,,M,5.0,N,9.3,K*5A
$PCDIN,01F112,0019FB9E,14,FF5C3DFF7FFF7FFC*54
$GPHDT,90.0,T*0C
$PCDIN,01FD02,0019FBA9,14,FF040359D302FFFF*5E
$GPMWV,310.0,R,7.7,M,A*2B
$GPMWV,286.5,T,6.2,M,A*22
$GPMWD,286.5,T,286.5,M,12.0,N,6.2,M*64
```

- `VTG` COG/SOG = 73.9°/6.2kn :heavy_check_mark: 
- `VHW` HDT/STW = 90°/5kn :heavy_check_mark: 
- `HDT` 90° :heavy_check_mark: 
- `MWV-R` AWA/AWS = 310°/7.7m/s :heavy_check_mark: 
- `MWV-T` TWA/TWS = 286.5°/6.2m/s :x: should be 292° 12.4kn (6.4m/s) because true wind is relative to water and must be calculated from HDT/STW, here COG/SOG is used and yields ground wind instead
- `MWD` :x: same problem as in 1. (angle instead of direction)
- SET/DFT are not calculated or converted (PGN 130577)

## Workaround

Blacklisting `MWD`, but `MWV-T` is still wrong.

## Suggestions

- The calculation should be fixed like it is done in the [Sail_Instrument](https://github.com/kdschmidt1/Sail_Instrument).
- Set and drift should be calculated and/or converted as well.
- There should be a config option to turn off all calculations, so only conversion is performed by the gateway and the calculations are done by downstream software. (It is not possible to just blacklist calculated sentences, `MWV` is sent in two flavors which cannot be blocked separately.)

wellenvogel, 2024-08-20:
I agree...
Mainly old code there.
And switching of computations is for sure a good idea.

quantenschaum, 2024-08-21:
To make the converter work correctly with systems that send wrong wind data (i.e. true wind in 130306-True_North [ground wind]) it would make sense to have a config option to select how the data in PGN130306 is interpreted. Like 2 pairs of dropdown boxes N2K <-> 0183 with the following options

| N2K | 0183 |
|---|---|
| 0 true north ground dir | AWA/S |
| 1 magnetic ground dir | TWA/S |
| 2 apparent angle | TWD/S |
| 3 ground angle | GWA/S |
| 4 true angle | GWD/S |

Then you could select something like

- `2 apparent angle <--> AWA/S`
- `0 true north ground dir <--> TWD/S`

which tell the GW to interpret the data.
 
This way you can tell the GW to do the conversion right even if your devices on the bus send the data with the wrong reference flags.

wellenvogel, 2024-10-07:
I started to implement the proposed mappings for the N2K wind reference.
As far as I see we currently can only use 3 values at the NMEA0183 side: AWA/S,TWA/S,TWD/S.
For the others we would need to implement more computations - similar to  [Sail_Instrument](https://github.com/kdschmidt1/Sail_Instrument).
Finally I'm not really sure if this would make sense in the gateway...
@quantenschaum: could you imagine to write a short summary for the wind definitions and issues that we could directly add to the documentation? Maybe just a wind.md in the doc directory?
Currently the information is a bit distributed and potentially hard to follow for the normal user.

quantenschaum, 2024-10-08:
That sounds good!

>could you imagine to write a short summary for the wind definitions
>Currently the information is a bit distributed and potentially hard to follow

That's true. Sure, I could wrap this up. I would merge the information from

- https://github.com/quantenschaum/Sail_Instrument/blob/master/docs/3-Calculated-Data.md#wind
- https://github.com/wellenvogel/avnav/issues/367
- https://github.com/SignalK/specification/issues/520#issuecomment-2304288433
- https://github.com/wellenvogel/avnav/issues/320#issuecomment-1931770688
- https://github.com/SignalK/signalk-derived-data/issues/116
- https://github.com/SignalK/signalk-derived-data/issues/75
- https://github.com/SignalK/signalk-derived-data/issues/5

>Finally I'm not really sure if this would make sense in the gateway

Performing calculations in the gateway can make sense in certain cases when the devices receiving the data need these fields but are not capable of calculating them on their own. But it should be optional. Performing the calculations downstream in the receiving devices/software might be more flexible.

wellenvogel, 2024-10-08:
> Performing calculations in the gateway can make sense in certain cases when the devices receiving the data need these fields but are not capable of calculating them on their own. But it should be [...]

Completely agree.

wellenvogel, 2024-11-14:
first steps with https://github.com/wellenvogel/esp32-nmea2000/releases/tag/20241114

skrue, 2025-07-04:
Hi, what's the current status of the true wind calculation? Any way to use it with [20250305](https://github.com/wellenvogel/esp32-nmea2000/releases/tag/20250305)? The release log for [20241114](https://github.com/wellenvogel/esp32-nmea2000/releases/tag/20241114) mentions `remove invalid true wind calc, allow to configure some mapping - partly #75` but no further details. I looked into https://github.com/wellenvogel/esp32-nmea2000/pull/76/commits/54d6d51519a65c8cab6af187a8286d85b5ded2a1 and tried to understand what has changed and it seems that commenting out lines `506` ff. essentially disabled TWS/TWA calculation.

I am currently using the gateway for NMEA2K to WiFi only. My instruments are Simrad IS12, so running on SimNet, which I have physically converted to NMEA2K. The wind vane outputs AWS+AWA directly to the network and the IS12 Wind instrument then calculates and displays TWS+TWD. Unfortunately it does not feed the calculated values back into the network though, which is why the option to have the calculations done in the gateway would come in very handy.