wellenvogel/esp32-nmea2000 #77: Stop access point time not working

norbert-walter, 2024-08-30:
When  I using a stopAPTime grater than 0 the AP will not stop. I see allways the WiFi SSID on air. The device is not connected with anything other devices. Also not as WiFi client.

free-x, 2024-08-30:
I can partly confirm this behavior. After set to 5 minutes, AP becomes "hidden" flag, but AP is not really stopped

wellenvogel, 2024-08-30:
I cannot confirm this.
Which version did you run?
I tested it using the latest version (dev build) 20240429 - directly on an M5-Atom with CAN base, setting to 2 minutes.
Did not run Wifi scans...
When using logging you can see something like:

LOG:1814:GWWIFI: AP auto shutdown enabled (120s)
....
GWSERIAL:121815:GWWIFI: shutdown AP

And afterwards the AP cannot be found any more when scanning wifi.

norbert-walter, 2024-08-30:
Thats the last build on OBP60 and not on a M5Stack.

wellenvogel, 2024-08-30:
In the UI you set minutes ;-).
Can you provide the logs? Especially the first line about auto shutdown enabled gives you the time.
Most probably I will add a log to write out the number of connected clients - I would suspect that there is still some client being connected.

norbert-walter, 2024-08-30:
Ohh shit. My problem. Sorry for the misstake.

closed 2024-08-31 by wellenvogel