wellenvogel/esp32-nmea2000 #78: No WiFi login possible after firmware update via web interface

norbert-walter, 2024-09-03:
After a firmware update via the OBP60 website, you can no longer log into the WiFi of the OBP60 access point even though an SSID is visible. For now, the only way to get around this is to integrate the OBP60 into the home network as a WiFi client. The client connection remains if you do the normal update, which retains the settings. Then you can continue to access the OBP60 via the home network. You can only get access via the access point if you flash via USB.

wellenvogel, 2024-09-04:
Would be good to know about the environment - i.e. which software was installed before, which settings have been there and which software was installed.
Did you try this also with the base software - i.e. without the OBP60 software?

norbert-walter, 2024-09-05:
The environment is the OBP60 with the software  20240823_Test.

wellenvogel, 2024-09-22:
Was only related to OBP60 (Problem when updating the framework version), refer to https://www.segeln-forum.de/thread/73307-multifunktionsdisplay-diy/?postID=2685971#post2685971, https://www.segeln-forum.de/thread/73307-multifunktionsdisplay-diy/?postID=2686100#post2686100

closed 2024-09-22 by wellenvogel