wellenvogel/esp32-nmea2000 #79: Support for UDP broadcast

JSC-Starcat, 2024-09-25:
One (i.m.o.) reasonably easy to add useful feature would be NMEA broadcast over WiFi to UDP port 10110 (which seems to be the standard for this) or a different port. Lots of software supports this: OpenCPN, TZ iBoat among others.

I've seen very simple implementations that send NMEA messages to the broadcast address of the current subnet (i.e. when the NMEA gateway and the client are on subnet 192.168.10.x, the broadcast address would be 192.168.10.255) and these would be picked up by any client on the subnet listening on that particular UDP port.

Advantages would be the minimum amount of configuration neccessary and the fact that UDP is a very lightweight protocol.

wellenvogel, 2024-09-26:
Fur sure a nice idea.
I will have a look at it. The only issue:
The IP stuff uses a lot of the processor power...
And at least some decisions would be necessary about the destination addresses (at the end you maybe even would need 2 channels - one for the Access Point IP, one for the WifiClient).
Not sure if multicast (not broadcast) should also be an option.

wellenvogel, 2024-11-14:
https://github.com/wellenvogel/esp32-nmea2000/releases/tag/20241114

closed 2024-11-14 by wellenvogel