wellenvogel/esp32-nmea2000 #80: actisense nmea reader timeout

CGZ127, 2024-10-01:
Hello,
    I'm using ATMO LITE+ATOM CANBUS,flash m5stack-atom-20240428-all.bin,and change usb mode to actisense, but using the nmea reader software the log shows the get operating mode timeout, I wonder if anything wrong or the board chip changes, could you help me find out?Thanks!
[image]
[image]
[image]

CGZ127, 2024-10-01:
I have tried，the default baudrate is 115200 after flashing ,not working，I've tried from 4800 to 115200, none of them can work

---- Replied Message ----
| From | ***@***.***> |
| Date | 10/01/2024 17:42 |
| To | ***@***.***> |
| Cc | ***@***.***>***@***.***> |
| Subject | Re: [wellenvogel/esp32-nmea2000] actisense nmea reader timeout (Issue #80) |

At First, Set baudrate to 115200 or 230400

Regards
free-x

wellenvogel, 2024-10-01:
I don't think that you can use any actisense tool to communicate.
Although data will be send on the USB port in actisense format the device does not support the full actisense Protocol.
Especially it will not receive/send any of the control messages.
Using it with other Software that reads/sends data in actisense format should be fine.

CGZ127, 2024-10-03:
hello again, I just got a usb2can, I was trying sending a HDT sentence by serial to atom ,but nothing can received by the usb2can,and the monitor webpage shows nmea2000 offline, the serial debug dialog shows nvs_open failed, I don't know if this is the problem
[image]
[image]
[image]
[image]

wellenvogel, 2024-10-03:
For sure you miss a ground connection and bus termination resistors.
But I have no idea on usb2can - i.e. which software you use to connect to it, which protocols it supports...

CGZ127, 2024-10-03:
I had connected the gnd cable and the 120r terminal resistor is built in the usb2can，the software is pcan view，a canbus monitor software can catch & display raw data on canbus，if the error shows in the debug dialog doesn't matter, I'm considering if the 3035 can bus transceiver is broken.....

---- Replied Message ----
| From | Andreas ***@***.***> |
| Date | 10/03/2024 17:47 |
| To | ***@***.***> |
| Cc | ***@***.***>***@***.***> |
| Subject | Re: [wellenvogel/esp32-nmea2000] actisense nmea reader timeout (Issue #80) |

For sure you miss a ground connection and bus termination resistors.
But I have no idea on usb2can - i.e. which software you use to connect to it, which protocols it supports...

wellenvogel, 2024-10-03:
The log about nvs should not matter.

wellenvogel, 2024-11-13:
As there is no further input I will close the issue.

closed 2024-11-13 by wellenvogel