wellenvogel/esp32-nmea2000 #81: OBP60 dev20241006, USB no longer usable after a while

norbert-walter, 2024-10-06:
If an output via USB has been activated, the output freezes after a while. The serail speed is 230400 Bd. Logging is disabled and only 8 NMEA0183 sentences are transmitted per second. If a white filter is used for exactly one sentence, the output does not freeze. However, the data transfer is very jerky. It looks as if the data processing for USB is not running fast enough. The same problem is when I use the actisense mode. With all debug information comming the error faster after any seconds.

wellenvogel, 2024-10-06:
Can you try again with 115200?
Never had this issue.
What exactly is freezing?
Only the USB output?

And: Which version...

norbert-walter, 2024-10-06:
With 115200 is the same situation. Freezing is complete stop of transmission on the USB output. Version see headline.

thooge, 2024-10-16:
Tried with 57600 same freeze here.
NMEA to USB off
NMEA from USB off
USB to NMEA2000 off
log level = debug

wellenvogel, 2024-10-16:
Is this also on OBP60?
We already found that there are some issues especially for the S3 used in the OBP.
Seems that after some time the USB driver thinks that it cannot deliver the data - and thus after a timeout the data is dropped.
Norbert was going to test a couple of things to maybe get closer there.

thooge, 2024-10-16:
Yes this is in OBP60. Thought commenting a OPB60 issue.
After changing log level away from debug it seems to work better.

W-Geronius, 2024-10-16:
same here, OBP60 V2. Log level 'debug' ...

wellenvogel, 2024-11-14:
https://github.com/wellenvogel/esp32-nmea2000/releases/tag/20241114

closed 2024-11-14 by wellenvogel