wellenvogel/esp32-nmea2000 #82: Implementation of NMEA2000 RAW streams to WIFI

torbenmb, 2024-10-10:
Has it already been considered to also implement one of the open protocols to stream NMEA2000 RAW data to WIFI?

In addition to streaming the NMEA2000 data in actisense format via USB, it would be great to also stream them wirelessly in one of the following open formats:

* Yacht Devices RAW (https://www.yachtd.com/products/wifi_gateway.html)
* Digital Yacht RAW  (https://github.com/digitalyacht/iKonvert/wiki/4.-Serial-Protocol)

Kind regards,
Torben

wellenvogel, 2024-10-10:
Not yet...
Currently in supports SeaSmart.
For the others someone :-) would need to write the code to translate.

zajac-grzegorz, 2025-01-01:
I have recently created my variation of the Nmea2000->Wifi gateway, and it has multiple converters to provide the output on TCP or UDP asynchronous servers in the following formats:

Actisense NGT, BIN, RAW and ASCII, Yacht Devices ASCII RAW, SeaSmart, NMEA0183 and in the custom JSON format. Support for fast packet messages for RAW formats included.

Feel free to check or reuse the code.

https://github.com/zajac-grzegorz/ReSailing-NMEA2000-Gateway

wellenvogel, 2025-01-05:
Thanks, looks interesting!

thomasja27, 2026-02-26:
Thank you for all the work you have put into this project. I am also looking for this option (Yacht Devices ASCII RAW over WIFI), I am trying to get data from batteries and engine data from Signal K onto the NMEA2000 network so I can display this data on the Garmin chartplotter, but it seems that the Signal K to NMEA2000 plugin only supports NMEA2000 connections and not NMEA0183. Maybe this can be solved in another way, but my knowledge stops here.

thomasja27, 2026-02-26:
The problem is that the plugin for sending sensor data to NMEA2000 in Signal K only talks to NMEA 2000 connections in Signal K. I have got the gateway working and getting data into Signal K via NMEA 0183, but I have not been able to send data that I have in Signal K back to the NMEA 2000 network. I have a fuel gauge, tachometer, oil pressure and engine temperature that I have integrated into Signal K via an ESP. I am now trying to get this sensor data sent over to the NMEA 2000 network to display the data on the plotter.