wellenvogel/esp32-nmea2000 #86: preparing firmware for ESP32-C3 supermini

burbon-pl, 2024-10-11:
Hi!
Is there any easy way to prepare your soft for different ESP32 boards? Like ESP32-C3 supermini for example? "Nodemcu-homberger" unfortunatelly doesn't booton this board. But it works just perfectly on ESP32-WROOM-32D.

wellenvogel, 2024-10-11:
Most probably it simply needs some additional board definitions.
If you want to play with it you could fork and go for a usertask: https://github.com/wellenvogel/esp32-nmea2000/blob/master/lib/exampletask/Readme.md.
So you could easily define a new environment with this board type. If you made some testing on it I could potentially include this in the core - and also for the online build service.
Or just directly go for a PR with the necessary board definition.

wellenvogel, 2025-03-04:
Are you still interested in this part? I will close the issue for now - feel free to reopen if you would be able to help - e.g. by testing.

TaaraLabs, 2025-03-12:
Hi, I am also having trouble compiling for esp32c3  (lolin_c3_mini). Interestingly the same error occurs with some esp32 boards  like wemos32d1mini even tough the MCU is the same as nodemcu-32s. At least the binary compiled for nodemcu-32s works on wemos32d1mini boars too. No surch solution for esp32c3 so far.
`Compiling .pio/build/lolin32c3mini/libff5/NMEA2000-library/N2kGroupFunctionDefaultHandlers.cpp.o
src/main.cpp: In member function 'virtual void DefaultLogWriter::write(const char*)':
src/main.cpp:747:13: error: 'USBSerial' was not declared in this scope
             USBSerial.print(data);
             ^~~~~~~~~
src/main.cpp:747:13: note: suggested alternative: 'GwSerial'
             USBSerial.print(data);
             ^~~~~~~~~
             GwSerial
Compiling .pio/build/lolin32c3mini/libff5/NMEA2000-library/N2kMaretron.cpp.o
*** [.pio/build/lolin32c3mini/src/main.cpp.o] Error 1`

wellenvogel, 2025-03-12:
Refer to https://github.com/wellenvogel/esp32-nmea2000/blob/master/doc/serial-usb.md - you need a define for USBSerial for your board.
Maybe just fork and use the extension approach (https://github.com/wellenvogel/esp32-nmea2000?tab=readme-ov-file#extending-the-software).
With an own platformio.ini you can easily add an environment and the necessary define.

TaaraLabs, 2025-03-12:
`#define USBSerial Serial` helped for wemos_d1_mini32 board, thank you!
lolin_c3_mini still does not compile, complaining about Serial2 now. Tried everything (i think) serial-usb.md doc suggested.

`lib/channel/GwChannelList.cpp:321:50: error: 'Serial2' was not declared in this scope
             serialStream=createSerial(streamLog,&Serial2,param->id);
                                                  ^~~~~~~
lib/channel/GwChannelList.cpp:321:50: note: suggested alternative: 'Serial1'
             serialStream=createSerial(streamLog,&Serial2,param->id);
                                                  ^~~~~~~
                                                  Serial1
Compiling .pio/build/lolin32c3mini/lib2e5/config/GWConfig.cpp.o
*** [.pio/build/lolin32c3mini/lib8b7/channel/GwChannelList.cpp.o] Error 1`

My fork and exension is [here](https://github.com/TaaraLabs/esp32-nmea2000/tree/master/lib/taaralabstask).

wellenvogel, 2025-03-12:
Ok, it seems that the C3 only has 2 UARTs.
The code is not really prepared for this.
I will have a look.
In the meantime you could simply use
#define Serial2 Serial1
for the C3.
This should help to compile.
The code for Serial2 should only run if you configure a second serial - what is not the case.

TaaraLabs, 2025-03-12:
`#define Serial2 Serial1` got rid of the Serial2 problem but it seems that C3 differs in even more ways:

`In file included from lib/spitask/GwSpiTask.cpp:16:
lib/spitask/GwSpiSensor.h:140:21: error: 'SPI3_HOST' was not declared in this scope
 #define GWSPI2_HOST SPI3_HOST
                     ^~~~~~~~~`

wellenvogel, 2025-03-13:
The same issue again.
Could be solved the same way.
#define SPI3_HOST SPI2_HOST
Not really great but as a workaround.

TaaraLabs, 2025-03-13:
`#define SPI3_HOST SPI2_HOST` gives the following error:
`lib/spitask/GwSpiTask.cpp: In function 'void runSpiTask(GwApi*)':
lib/spitask/GwSpiTask.cpp:78:13: error: duplicate case value
             case GWSPI2_HOST:
             ^~~~
lib/spitask/GwSpiTask.cpp:68:13: note: previously used here
             case GWSPI1_HOST:
             ^~~~
*** [.pio/build/lolin32c3mini/lib8bc/spitask/GwSpiTask.cpp.o] Error 1`

It does compile with `#define SPI3_HOST SPI1_HOST`. Even N2K from CAN seem to work but saving settings on the web interface does not work so it's impossible to configure Serial port speed for the RS485 sensor.

wellenvogel, 2025-03-13:
Ok, so the workaround should be ok.
What happens when you try to save the settings?
Maybe you could just connect via USB and get the log when trying to save.

TaaraLabs, 2025-03-13:
After changing serial baud rate and pressing Save&Restart button the UI restored the old value. However I discovered that after power off/power on and ReloadConfig the new value was present and activated. Another problem is that esp32c3 would not boot with RX/TX pins I was using. Changing them to default RX/TX(20/21) it started working in RX and UNI mode, not BI mode. There was also something weird happening between serial and USB. Status page showed data coming from the USB even when there was nothing connected to USB port. I just turned all three options on the usb port tab to off. Also I had to change USBSerial to Serial1 instead of Serial to get Serial working at all. Perhaps there could some more investigating and testing done but unfortunately my only RS485 NMEA0183 sensor just died and I cannot use it for testing anymore. Hope that the warranty is still valid.
At least with the following settings both CAN and RS485 work on ESP32C3. No USB functionality though. I also updated this info on my github fork.
My hope was that I can use Lolin ESP32C3 mini with the same pinout as Wemos D1 Mini32 but no such luck this time.
Thank you for the very fast and helpful replies and overall awesome support!
```
#define ESP32_CAN_TX_PIN 1
#define ESP32_CAN_RX_PIN 0
#define GWSERIAL_RX 20
#define GWSERIAL_TX 21
#define GWSERIAL_TYPE GWSERIAL_TYPE_RX
#define ARDUINO_USB_CDC_ON_BOOT 1
#define USBSerial Serial1
#define Serial2 Serial1
#define SPI3_HOST SPI1_HOST
```