wellenvogel/esp32-nmea2000 #89: OTA Firmware full Update failing: missing magic byte

W-Geronius, 2024-10-19:
Updating the firmware by selecting a xxx-all.bin always fails displaying an error 'missing magic byte at ...'
Updating the firmware by selecting a xxx-upate.bin never fails.

 -all.bin file apears to be checked like a -update.bin file (example using norbert-walter/esp32-nmea2000-obp60/releases/download/20241006/obp60_s3-dev20241006-all.bin)

[image]

At position 288 (0x0120) of a xxx-all.bin file the magix byte can't be found - checkImage should inspect position 65824 (0x010120)
All magic bytes start there:
[image]

wellenvogel, 2024-10-19:
You cannot flash an ...all.bin with OTA (what would be the use case for it?).
This is only intended for initial flash via the WebFlasher or esptool.

W-Geronius, 2024-10-19:
Ok, that wasn't clear (to me) from the documentation. I will suggest to include that hint in Norbert's document.
The issue may be closed, thanks

closed 2024-11-04 by wellenvogel