wellenvogel/esp32-nmea2000 #102: reconnect to raspberry pi 4B (AvNav bookworm lite) fails

W-Geronius, 2025-02-18:
All devices based on the nmea2000-gateway (various OBPxx displays, but also M5 Atom lite) can - once hard resetted - no longer connect to the wlan-ap of the AvNav-Raspi (image_2024-09-07-avnav-bookworm-armhf-lite.zip with current updates). This failure only applies to bookworm powered AvNav Servers.

ESP log via USB:

>[2025-02-11 11:43:31] GWSERIAL:94919:wifiClient: retry connect to B_OP_20101
[2025-02-11 11:43:31] GWSERIAL:94925:creating wifiClient ssid=B_OP_20101
[2025-02-11 11:43:31] GWSERIAL:94930:wifiClient connect returns 4

to reproduce:

- gateway: Atom running FW 28.11.2024 (alternatively **any** OBPxx V1 or V2) 
- host: Raspi 4B running image_2024-09-07-avnav-bookworm-armhf-lite or image_2025-02-03-avnav-bookworm-armhf-lite.zip
- host: (AvNav) creates wlan-ap 
- gateway: accesses host via WiFi, 
- gateway: TCP CLient reads AvNav nmea0183 data via port 10110 

Initial connection works, data arrive as expected 

Error: 
- gateway can't reestablish WIFi connection once hard resetted (power off). 
Rebooting the host or executing ifdown wlan-ap && ifup wlan-ap will fix the situation

Error does not occur: 
- if the gateway performs a soft reset (via the web interface or by crashing) 
- if the host runs a different operating system, e.g. Android or OP image (Buster) 
- on first connect after host restart

[image] [image]

[\[com COM4\]  (2025-02-12_220139)  _COM04 Atom.log](https://github.com/user-attachments/files/18840623/com.COM4.2025-02-12_220139._COM04.Atom.log)
[hostapd.log](https://github.com/user-attachments/files/18840624/hostapd.log)
[journalctl.log](https://github.com/user-attachments/files/18840626/journalctl.log)
[tcpdump.pcap.log](https://github.com/user-attachments/files/18840625/tcpdump.pcap.log)

wellenvogel, 2025-03-04:
Basically this is caused by a firmware update for the Wifi chip of the pi when updating from bullseye to bookworm.
I created an issue in the [firmware repo](https://github.com/RPi-Distro/firmware-nonfree/issues/49) - but I'm not sure if there is any chance for a response/change there.
In the meantime there are 2 workarounds/solutions:
(1) On the pi you can switch the firmware of the Wifi chip (see linked issue):
`update-alternatives --set cyfmac43455-sdio.bin /lib/firmware/cypress/cyfmac43455-sdio-minimal.bin`
Afterwards you need to restart the Pi.
The drawback is that we do not know if a next firmware update will break this again on the pi.
(2) To be able to reconnect the ESP needs to be "silent" for a certain time so that the Pi will detect it as "inactive" and will remove it from the list of know stations. This time by default is 300s (i.e. 5 minutes) - parameter ap_max_inactivity in the [hostapd configuration](https://web.mit.edu/freebsd/head/contrib/wpa/hostapd/hostapd.conf). Therefor the reconnect-tries of the ESP must be larger then this time (or the ESP needs to be switched of for at least this time). As 5 minutes is not really acceptable we need to change the ap_max_inactivity on the raspberry to e.g. 30s. Additionally there will be a change on the ESP side to only reconnect after 40 if the reconnect fails a couple of times after the start.

wellenvogel, 2026-08-19:
20250305

closed 2026-08-19 by wellenvogel