wellenvogel/esp32-nmea2000 #107: DBT message not processed nor converted

Scorgan01, 2025-04-14:
When receiving messages of type DBT, they are not displayed on the 'Data' page of the web interface, nor are they converted to N2k messages.
They are listed, however, on the 'Statistics' page of the web interface, and are forwarded as NMEA0183 messages to a TCP client. The issue happens regardless of the source channel they have been received, either Serial or TCP.
In the attached debug log you can see, that messages of type DBT have been received. When I create a faulty message (here: checksum error), this is recorded in the log like this:
`// DBT message with wrong checksum`
`GWSERIAL:610749:NMEA[5]: $IIDBT,19.3,f,5.88,M,3.22,F*1C`
`GWSERIAL:610749:NMEA0183DataToN2K[5] invalid message $IIDBT,19.3,f,5.88,M,3.22,F*1C`

Obviously, the message is processed somehow in the module 'NMEA0183DataToN2K'.

BR, Uli

[NMEA0183.txt](https://github.com/user-attachments/files/19735816/NMEA0183.txt)

Scorgan01, 2025-04-18:
Today, I recognized that, even though a DBT message is not converted to N2k, it can be forwarded as an N0183 message via TCP connection. The gateway will still not display the data on its data web page, but AvNav shows the data as DPT.

wellenvogel, 2025-04-21:
The forwarding of NMEA0183 messages is completely independent of the translation to N2K. I.e. all NMEA0183 messages are always send out on the NMEA0183 interfaces (if not blocked by filters).

Scorgan01, 2025-04-26:
Thanks for that correction.
I just compiled a version and checked it at my desk with some simulation data and here it works. The OBP60 display also shows proper DBT values after I applied the changes to that code.
 I can check it in a week on the boat in real life.

Are you sure, however, about line 679 in <NMEA0183DataToN2K.cpp>:
`  SetN2kWaterDepth(n2kMsg,1,dbs,offset); //on the N2K side we always have depth below transducer` ?
I am wondering if it should read
`  SetN2kWaterDepth(n2kMsg,1,dbt,offset); //on the N2K side we always have depth below transducer` ?

Scorgan01, 2025-04-28:
Oh, alright, should have analysed the whole function before I did this comment. Anyway, thanks for the fix.