wellenvogel/esp32-nmea2000 #109: 20250305 redefinition errors in generated/GwConfigDefinitions.h

W-Geronius, 2025-06-10:
After importing the latest gateway code into my OBP60 repository, I noticed errors when building env OBP60_s3 and OBP40_s3.
The OBP60 build succeeds, but OBP40_s3 fails due to a myriad of errors like:

> In file included from lib/config/GWConfig.h:6,
from lib/api/GwApi.h:6,
from src/main.cpp:22:
lib/generated/GwConfigDefinitions.h:434:43: error: redeclaration of 'constexpr const char* const GwConfigDefinitions::deviceName'
static constexpr const char* deviceName="deviceName";
^~~~~~~~~~~~

This might well have happened because of incorrect adaptation of platformIO to the new requirements (which are unclear to me) - or otherwise require a check in the generation process. 

To fix this, claude suggested to modify mergeConfig() in extra_script.py to track already processed config items. 
This fix seems to work - see code suggested:

`              # modify mergeConfig() in extra_script.py
                ...
                merge=json.load(ah)

                # Create a dictionary to track existing items by name
                existing_items = {}
                for item in base:
                    name = item.get('name')
                    if name:
                        existing_items[name] = item
                
                # Add only non-duplicate items from the merge config
                for item in merge:
                    name = item.get('name')
                    if name and name in existing_items:
                        print(f"Warning: Skipping duplicate config item '{name}' from {cname}")
                    else:
                        base.append(item)
    return base
`

closed 2025-06-10 by W-Geronius

wellenvogel, 2025-06-11:
Yes.
Just do this for both environments - i.e. distinct names for each of them.
Just what I already asked @norbert-walter to do.
You could even have a common part in a config.json...
https://github.com/wellenvogel/esp32-nmea2000/blob/master/lib/exampletask/Readme.md

W-Geronius, 2025-06-11:
Yes, I should have read the readme with more attention. Sorry. 
On the other hand exploring via code holds the benefit of better understanding 
Now I have both env clearly separated in my repo. 

I also considered to split into a general config.json (common part) and the distinct entries within the custom configs - that would be significantly easier to maintain. It seems rather impossible however, as the omnipresent capabilities attribute always differ between obp60 and obp40 - I couldn't figure out its purpose yet, but it looks like a crucial thing throughout the application